UHHRobotics22-23/MarimbaBot #42: Collect initial handwritten whiteboard data set

Juphex, 2023-04-13:
500 (human) handwritten notation on the whiteboard for 100 automatically generated note sequences (5 images per sequence)

dataset storage: https://cloud.mafiasi.de/f/17859573

data collection script: https://github.com/UHHRobotics2223/robot_project/blob/main/marimbabot_vision/scripts/real_world_data_collection.py

Juphex, 2023-07-13:
@Flova maybe you give an update about the importance of this issues for training

@ateRstones and I wondered

Flova, 2023-07-14:
The extended model is currently training. There were a few issues in the beginning, as the extended dataset was not formatted consistently, also artificial data needed to be generated and we needed a more complex set of tokens. Let's see how this first run turns out. We might also want to try character level tokenization. Also there seem to be articulations in the labels of the whiteboard data but they are disabled in the data generation, can you give any insides on this?

Flova, 2023-10-16:
Out of scope for now