UHHRobotics22-23/MarimbaBot #43: Collect extended data set

Juphex, 2023-04-13:
add accidentals, different note durations, 'dotted' notes ... 

dataset storage: https://cloud.mafiasi.de/f/17816624

data collection script: https://github.com/UHHRobotics2223/robot_project/blob/main/marimbabot_vision/scripts/real_world_data_collection.py

Juphex, 2023-05-02:
I would like to tackle this task this week. Can anyone join me? @ateRstones @Flova @iibrahimli  @cvhex @yunlong-wang-cn @berkgungor @scaredycode

Flova, 2023-05-02:
I will keep working on the motion side of things, but feel free come to me if you have any additional questions.

yunlong-wang-cn, 2023-05-02:
@Juphex  Even though I also have some stuff to do, I will join you.

Juphex, 2023-05-04:
I had to add slurs, accidentals and dynamics to the data set first. We, or other people, could do it next week @yunlong-wang-cn :)

yunlong-wang-cn, 2023-05-04:
when you need help, just let me know. @Juphex

Juphex, 2023-05-31:
@Flova 
Since this is a time-exhaustive task and we still have (https://github.com/UHHRobotics22-23/marimbabot/issues/42) open : somewhat like 30 samples should be enough, right?

Juphex, 2023-06-19:
@Flova  we have 16 samples now. Would you like to try it with 16 real-life samples right now? I mean we can apply data augmentation on them 😅

Juphex, 2023-08-10:
As discussed today in our meeting,
the real dife data collection is on hold and waits for the results of training on handwritten data generation by @scaredycode .

closed 2023-10-16 by Flova