UHHRobotics22-23/MarimbaBot #51: Module integration

scaredycode, 2023-04-13:
integrating modules according to user story

Juphex, 2023-04-13:
User story:

 1. User add melodic with megnic to white board, and robot plat that. (lilypond)
 2. voice command to robot, commands include [read, start, stop, repeat, concatenate, slower, faster, louder, softer]
 3. Test to speech, for hurman robot interaction, not for debug.
 4. Excution of the command

Flova, 2023-04-13:
We should add a ros package called `marimbabot_behavior` for the state machine / behavior tree / ... (what ever we want to use)

yunlong-wang-cn, 2023-04-13:
After the user story, I think we have some new component of the system(the following topic name just as example):
1. Speech recognition
     Recognize the speech signal from user and publish it to a ros topic  /user_speech .
2. Speech synthesis.
     Listening the message in the ros topic /speech_synthesis, and speak it out. 
3. High level control. (marimbabot_behavior).
     Listening the message in the ros topic /user_speech and response for that.

scaredycode, 2023-04-20:
see https://github.com/UHHRobotics22-23/marimbabot/issues/53 for topic (& communication) description

v4hn, 2023-04-21:
> I assume it just means "shifting" the notes by #,b or full notes. https://en.wikipedia.org/wiki/Transposition_(music) 

Transposition is *one* example, but not the only one meant here. We require that you can adapt different parameters that show that the system has a model of music theory (you get a lot of that for free with lilypond). Otherwise why would you have bothered with musical notation in the first place? Here are a number of suggestions, but there are lots more and please come up with whatever interests you:

- With Transposition you can create melodies that use the marimbaphone bars that you cannot detect in vision right now (I believe the whole upper octave)
- [Mode Changes](https://en.wikipedia.org/wiki/Mode_(music)) are different from transposition because the distance between individual notes in the scale changes (the position of the half-tone steps in the diatonic scale) and not all notes together. The classic difference is major to minor scale ("happy" to "sad" mood), but everything else is just as easy when you are there already.
- [Time signature](https://en.wikipedia.org/wiki/Time_signature) changes. This is very open-ended and there are any number of ways of creating melodies in one time signature that were originally in a different one, e.g., moving between `4/4`,  `3/4`, `2/4`, `6/8`, (`5/4`???)
- Style-changes: Many musical styles are based around specific rhythms, e.g., there is the classic [tango rhythm](https://en.wikipedia.org/wiki/Tango#/media/File:Tango_dance_pattern.png), so any melody you convert to such a rhythm  (yes, there are many ways to do that) will sound closer to Tango. I wouldn't be surprised if there are multiple projects around that can move melodies/pieces between styles.
- [Swing](https://en.wikipedia.org/wiki/Swing_(jazz_performance_style)) is another obvious idea here. `<biased opinion>`Everything sounds better when you swing it`</biased_opinion>` :-)
- With the two-mallet setup you could decide the notes to play with the second mallet based on harmonization, e.g., [improvise a second voice](https://www.reddit.com/r/musictheory/comments/9ax2y6/what_exactly_is_in_singing_second_voice_or/)
- change loudness in different ways.
  - overall loudness of the piece
  - [Syncopation](https://en.wikipedia.org/wiki/Syncopation) (yes, "stressing" might imply more than loudness changes)
  - add [crescendi/decrecendi](https://en.wikipedia.org/wiki/Crescendo) to a phrase.

In the end these commands can also help you to bring everything together in the integration even when not everything is supported throughout the pipeline (e.g., vision does not support addressing the upper octave or users don't have to specify 2-note-chords in the input to get both mallets to play something interesting).

@iibrahimli @berkgungor @cvhex for reference.

yunlong-wang-cn, 2023-04-23:
> I assume it just means "shifting" the notes by #,b or full notes. https://en.wikipedia.org/wiki/Transposition_(music) @yunlong-wang-cn

Based on my understanding, the possible text-form speech examples for "shifting" could be: "C four transpose" or "C sharp", is that right? @Juphex

Juphex, 2023-05-04:
@yunlong-wang-cn  sure, that would be possible

yunlong-wang-cn, 2023-08-15:
For the record here: 
Last day,  we try to input the command from speech recognition, and then we have the following findings:
1. The noise before the speech could influence command recognition.
2. For the long running time,  it seems some buffer gets stuck inside of whisper(the engine of speech recognition), I guess because the processing time is bigger than the processing time windows.

Therefore, consider to:
1. #225 Use keyword spotting to activate the whisper, only the keyword is detected, then the whisper with recognize the speech.
2. #223  Consider to using some denoising technic(optional, depend on testing result.)

Juphex, 2023-10-10:
The current status is that everything is integrated well.

ateRstones, 2023-10-12:
Maybe let's close this ticket?

closed 2023-10-14 by Flova