UHHRobotics22-23/MarimbaBot #73: create action server for planning

Juphex, 2023-04-20:
adds to https://github.com/UHHRobotics22-23/marimbabot/issues/50

Juphex, 2023-04-20:
could you give me some  goal, result, and feedback messages

@Flova  @scaredycode

Juphex, 2023-04-20:
idea: input > hit > feedback (possible/worked?)

Flova, 2023-04-21:
I think we have different levels. The behavior would call a motion action with the lilypond string, getting feedback on how much of the piece was played and if there was an issue/abort. On the motion side, this lilypond string is converted to a sequence of hit points (with timings). If this is done in a seperate node (as it seem to be the case currently) we would need to have an additional action forwarding hitpoint etc. to the planning/execution and passing the feedback from the execution back to the behavior. If the lilypond -> hitpoint part is a lib used in the motion only one action would be needed.

v4hn, 2023-04-21:
> The behavior would call a motion action with the lilypond string

I understand that this is what you have some code for right now, but be aware that this means that the code that generates the robot motions might need a rather sophisticated understanding of music theory/support for a wider range of lilypond syntax if you want to include some/any of the music-theory based transformations [mentioned in the integration](https://github.com/UHHRobotics22-23/marimbabot/issues/51#issuecomment-1517592716). Also, if you accept lilypond strings you want to be very concrete on what subset of the language you accept as valid input. Of course one way to handle this is to rely on a library to parse the string in the background into the datastructure you use to generate the motions, another would be to design an interface that only forwards such a datastructure (in the form of ROS messages) directly as input to the module that actuates the arm to hit bars.

This is essentially the traditional "do you factorize or build a monolith and do you do either in libraries or in ROS nodes" debate with upsides and downsides to either approach.
Essentially it affects
- how you can introspect the pipeline
- how you write unit tests
- how you can stub parts to test others in isolation
- how other components would hook in
  - for example I could imagine one module here that colors the target bars that need to be hit independent of the actual arm motion.

v4hn, 2023-04-26:
For reference, here is the documentation for what musical features are supported if you compile lilypond to midi at some point to generate playing instructions:

https://lilypond.org/doc/v2.23/Documentation/notation/supported-notation-for-midi
https://lilypond.org/doc/v2.23/Documentation/notation/enhancing-midi-output

closed 2023-06-01 by Juphex