UHHRobotics22-23/MarimbaBot #100: Investigate planning with ruckig

ateRstones, 2023-05-04:


Juphex, 2023-05-04:
Interesting. I don't know if that is similar to what you are looking for but I stumbeled accross https://github.com/petercorke/robotics-toolbox-python last year.

v4hn, 2023-05-11:
@Juphex That's a generic robotics library with various subsystems (similar to [MoveIt](http://moveit.ros.org), [Orocos](https://orocos.org/), [Drake](https://drake.mit.edu/), [tesseract](https://github.com/tesseract-robotics/tesseract). In general, any of them (and various others) can be used with different pros/cons.
Ruckig, in contrast, is a very specific algorithm for motion generation (*not* planning) that allows to specify times, positions, velocities and accelerations for waypoints/goals.
For hitting notes w.r.t. different rhythms and loudness, we expect that goal times and waypoint velocities could simplify this a lot.

ateRstones, 2023-05-11:
pykdl
moveit_commander (urdf loader)

Juphex, 2023-07-13:
@Flova Do you think this issue is still relevant?
@ateRstones  and I wondered

Flova, 2023-08-29:
Closing this in favor of #214