UHHRobotics22-23/MarimbaBot #103: Tune PID controller for UR5 Gazebo

iibrahimli, 2023-05-11:
Right now the arm controllers are unstable / have steady state error. PID values need tuning (in file `marimbabot_ur5_moveit_config/config/arm_controller_ur5.yaml`)

iibrahimli, 2023-05-11:
TODO: Norman suggested that we should try just adding the mallet to his setup (`tams_ur5_gazebo`) and see if the default PID values work in that case

iibrahimli, 2023-05-11:
UPD: The default PID values cause shaking for smaller movements (e.g. move pan by 0.1 rad) while being stable for long-range movements (e.g. from one predefined position to another). Thanks @normanhendrich for testing this with default UR5 setup.

Sensible next step is to test the motion planning scripts in Gazebo and tune the PID values based on that. I will get on that as soon as #80 is merged.

Juphex, 2023-07-13:
What is the current status?
@ateRstones  and I wondered

Flova, 2023-08-30:
I think they were tuned some time ago. If there are still issues open another ticket.

closed 2023-08-30 by Flova