UHHRobotics22-23/MarimbaBot #106: interface between command_node and behaviour_node

yunlong-wang-cn, 2023-05-11:
@scaredycode : 
We need to set up the interface between the command node and the behavior node.
Considering:
- the way to transport the message
- the message content to be transported
- all possible command
- other potential problems

yunlong-wang-cn, 2023-05-11:
currently, I have a proposal using the ros topic and the following message content:

std_msgs/Header header
string command

scaredycode, 2023-05-11:
Hi, right now, I am using a placeholder topic voice_recognition_node/recognized_command in the behavior node to which I can send some basic command strings for testing. I would probably suggest a name like commad_node/recognized_command.

We should definitely make a list of the commands we want to support.
Basic commands could be "read", "play", "stop", "repeat", "loop", ...
More tricky commands could be "play as 120 beats per minute", "play one octave higher", "play only the third bar", ...

Of course the user might not use this exact formulation, therefore, it would be nice if the command node could identify the associated command within the list of commands based on the speech recognition, and then send only this one to the behavior.

The message content also does not necessarily need to be a full sentence, but can of course also be an abstraction of the actual meaning behind it (e.g."play" -> "p", "play as 120 beats per minute" -> "p t=120").

closed 2023-06-19 by yunlong-wang-cn