UHHRobotics22-23/MarimbaBot #108: clarification on the "declaration" in the README and marimbabot_audio

Juphex, 2023-05-11:
What exactly is meant by this? Please rephrase https://github.com/UHHRobotics22-23/marimbabot/commit/c343914b9a805a56733aff134b848d7e56f5218b#r112913317
As this is from the same merge I guess:
- why is there a license in marimbabot_audio?
- please remove README in the same folder
- is an [__init__.py](https://github.com/UHHRobotics22-23/marimbabot/blob/main/marimbabot_audio/__init__.py) needed as it is not a python module in that sense?

Could you please clarify @yunlong-wang-cn ? As I assume that this should be changed :)

yunlong-wang-cn, 2023-05-11:
you are right, I'll correct those later on.

Juphex, 2023-05-12:
Thank you :) 
I guess this the declaration is refers to [note2midi.py](https://github.com/UHHRobotics22-23/marimbabot/blob/main/marimbabot_audio/scripts/note2midi.py)
and 
[onset_detection.py](https://github.com/UHHRobotics22-23/marimbabot/blob/main/marimbabot_audio/scripts/onset_detection.py), right?

Juphex, 2023-05-24:
Thank you @yunlong-wang-cn, we may adjust that 👍