UHHRobotics22-23/MarimbaBot #111: Create Interpreter Node

scaredycode, 2023-05-11:
Add interpreter node that is used to extract only the necessary information from a given lilypond string (obtained from the vision node) and a tempo (bpm).

For each note, the extracted information should include:
1. tone name + octave (e.g. c') -> String
2. starting time point in seconds (e.g. 0.0 for the first note) -> ros::Time (int)
3. tone duration in seconds (e.g. 0.5 for a quarter note and 120 bpm) -> ros::Duration
4. loudness level on a scale from 0 to 1 (e.g. 0.1 for ppp, 0.4 for mp, 0.6 for mf, 0.9 for ffff) -> double / float

The interpreter should subscribe to the behavior_node/read_notes topic to obtain the current lilypond string (vision_node/recognized_notes format) after the "read" command was issued.
The node should then publish the extracted information to a new topic (e.g. interpreter_node/extracted_info)

scaredycode, 2023-05-11:
```mermaid
flowchart TD
    R(Voice Recognition) --> |command| B{fa:fa-brain Behavior}
    V(Vision) --> |lilypond| B
    B --> |lilypond \n tempo| C(Interpreter)
    C --> |extracted_info| B
    B --> |extracted_info| D(Motion)
    B --> |extracted_info| E(Audio)
```

scaredycode, 2023-05-11:
Example:

input:
\tempo 4=60 \key f \major c'1 ( r1 ) c''1
[image]

expected output values (note, t_start, duration, loudness):
[ (c', 0.0, 4.0, 0.5), (c'', 8.0, 4.0, 0.5) ]

Flova, 2023-05-12:
Two comments:
- I think the octave should be included as an int and not as a part of the string describing the note. The note could also be described by a number, but I am fine with keeping it a string due to the better readability.
- It currently seems like you want to do the parsing in another node. A service would be better suited for that compared to a normal topic subscription, as the service is essentially a remote procedure call with a response making e.g. a blocking call way easier to implement. You might also consider skipping the ROS meaning you call it directly as a python module.

Juphex, 2023-05-12:
I don't know if you have done already, but I think it would be nice to include your visualizations somewhere in the repository/docs (in a README) :)

Juphex, 2023-05-25:
We would prefer to have cords as two note elements with the same start time instead of having a string like `<c' a'>`. The elements would be individual notes but they are at the same time.

Juphex, 2023-05-25:
New definition as discussed with @Flova, @Juphex and @scaredycode:

# 1. tone name (e.g. E#) -> String
# 2. octave number (e.g. 4) -> int
# 3. starting time point in seconds (e.g. 0.0 for the first note) -> ros::Time (int)
# 4. tone duration in seconds (e.g. 0.5 for a quarter note and 120 bpm) -> ros::Duration
# 5. loudness level on a scale from 0 to 1 (e.g. 0.1 for ppp, 0.4 for mp, 0.6 for mf, 0.9 for ffff) -> double / float

~ Flova

closed 2023-06-16 by scaredycode