UHHRobotics22-23/MarimbaBot #141: bug by speech sythesis

yunlong-wang-cn, 2023-06-08:
When I run `roslaunch marimbabot_speech test_speech_syntesis.launch`

Then a error reported as following:

... logging to /home/wang/.ros/log/03c9f788-0608-11ee-84e6-59ae4e0d4770/roslaunch-wang-Y7000P-99239.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://wang-Y7000P:44089/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    soundplay_node (sound_play/soundplay_node.py)
    speech_synthesis_node (marimbabot_speech/speech_synthesis.py)

ROS_MASTER_URI=http://localhost:11311

process[speech_synthesis_node-1]: started with pid [99264]
process[soundplay_node-2]: started with pid [99265]

**************************************************************
Error opening pygst. Is gstreamer installed?
**************************************************************

[soundplay_node-2] process has died [pid 99265, exit code 1, cmd /opt/ros/noetic/lib/sound_play/soundplay_node.py __name:=soundplay_node __log:=/home/wang/.ros/log/03c9f788-0608-11ee-84e6-59ae4e0d4770/soundplay_node-2.log].
log file: /home/wang/.ros/log/03c9f788-0608-11ee-84e6-59ae4e0d4770/soundplay_node-2*.log

yunlong-wang-cn, 2023-06-08:
@scaredycode  It said, `Error opening pygst. Is gstreamer installed?`. So how did you install the gstreamer and which version?

yunlong-wang-cn, 2023-06-08:
@scaredycode  I found the reason, you need to incorporate the sound_play package in your package.xml and CMakeLists.txt. Otherwise, the related software of sound_play will not be installed.

closed 2023-06-08 by yunlong-wang-cn

scaredycode, 2023-06-09:
Hi, sorry for the late reply and for the confusion. For the speech module, I use the soundplay node which has already been specified in the audio node (lilypond to audio). Thats why I did not want to specify it again in the speech package. 
I only launched soundplay in the test launch file because sound play will be launched by the audio module otherwise. Flo said launching the same node twice is bad practice, so I interpreted that as also keeping the soundplay node out of the other files as well, but listing it in CMakeLists.txt and package.xml does of course make a lot of sense. Thanks for the find, I will add it later today :)

yunlong-wang-cn, 2023-06-09:
@scaredycode Thanks for your reply. And I think how about merge speech and command recognition together, please take a look PR #142