UHHRobotics22-23/MarimbaBot #150: lilypond_to_audio - behavior integration

scaredycode, 2023-06-15:
create topic for lilypond_to_audio (message is lilypond string)

Juphex, 2023-06-21:
lilypond_to_audio already has a subscription on a topic. (Used outgoing from the vision network)
I will rename the topic that is subscribes to :)

Juphex, 2023-06-21:
There is already a subscribtion on a topic. 
However, the name is 'vision_node/recognized_notes'. Maybe it makes sense to create a second topic just exactly for the behavior part.

Flova, 2023-06-23:
The `vision_node/recognized_notes` is the vision output right? I think this issue was describing the `behavior -> midi synth` communication.

Juphex, 2023-06-23:
@Flova 
You are right. The audio not was just subscribing on it by default.

closed 2023-06-30 by Juphex