UHHRobotics22-23/MarimbaBot #153: possibly replace sound_play with audio_ play

scaredycode, 2023-06-15:
advantage: no tempfiles - just audio content, information about duration of audio
disadvantage: bad / no documentation, potentially difficult to stop sound from playing

could be used for speech synthesis and lilypond -> MIDI -> audio

Juphex, 2023-06-21:
That would affect
audio_from_lilypond.py
text2speech.py

if you want to change it over the whole project. I would welcome it :)

Juphex, 2023-06-22:
I assigned myself if that is okay with you @scaredycode  :)
Otherwise, please let me know.

Juphex, 2023-06-30:
I will take a look in the next days:)

You are right @scaredycode @yunlong-wang-cn , maybe "blocking" the python code is good enough since we use an action.

yunlong-wang-cn, 2023-07-01:
@Juphex If you need any help, just let me know

Juphex, 2023-07-05:
@scaredycode @yunlong-wang-cn 
I implemented it here https://github.com/UHHRobotics22-23/marimbabot/pull/172 .
Is that what we want? I also noticed that one cannot adjust the volume here.

Juphex, 2023-07-05:
@yunlong-wang-cn  yes, please, go ahead and take a look at audio_play or even implement it. As I have seen from the docs it subscribes to the topic "audio" and accepts the message type "audio_common_msgs::AudioDataConstPtr". I don't know if this is beneficial and if it makes things more complicated compared to what we have now. As this what we have already works appropriate😅

Juphex, 2023-07-06:
@scaredycode @yunlong-wang-cn 
As discussed, we already have the desired functions (e.g. like wait until an audio was played / abort / send audio) and they also work. It does not really make sense to switch because it could also create more work and our focus should be elsewhere.

closed 2023-07-06 by Juphex