UHHRobotics22-23/MarimbaBot #167: handle command detection during tts (or in complex/ loud environments)

scaredycode, 2023-06-29:


scaredycode, 2023-07-13:
It's basically how to handle voice recognition while the speech synthesis is actively producing speech or when people are speaking in the background. Yunlong said the command detection is very sensitive so this ticket is just a reminder to think about solutions in situations like this.
I think the keyword "Marimbabot" works fine but the command detection also needs 2s of silence to signal the end of a command, which might not always be possible in every situation.
So far no testing was performed and we might need to consider adjusting the detection sensitivity or applying some sort of noise filter.

And no, I'm not working on this. I opened the ticket for yunlong some time ago when I talked to him about it and he asked me to make a ticket.

Juphex, 2023-07-13:
Thank you for the elaboration on this. :)

yunlong-wang-cn, 2023-08-14:
I think it can be solved by adding [keyword spotting](https://github.com/zycv/awesome-keyword-spotting.git), which can extract text from speech when the keyword is spotted. And at the exhibition, the audience could focus on what you are doing, there could be not so much noise.

yunlong-wang-cn, 2023-10-03:
done in #230 by using hot word spotting, then close it.

closed 2023-10-03 by yunlong-wang-cn