UHHRobotics22-23/MarimbaBot #173: Incorperate the audio feedback into the planning evaluation and potential recalibration

ateRstones, 2023-07-05:


Juphex, 2023-07-13:
@scaredycode  @yunlong-wang-cn is that something you are currently doing?

scaredycode, 2023-07-13:
I think so far the Audio feedback only assigns one final score for the whole performance, which might not be enough for a fall recalibration. But @yunlong-wang-cn should probably be able to elaborate further.

yunlong-wang-cn, 2023-07-14:
@Juphex  In your opinion which part of the planning should be optimized? or recalibrated?  I don't know much about the details of motion planning, perhaps I need a brief introduction.

ateRstones, 2023-08-10:
We could use the audio feedback to detect when we missplayed and then use the information to adjust our world model of the marimba. But maybe this is not needed and a simple score suffices for us

yunlong-wang-cn, 2023-08-10:
> We could use the audio feedback to detect when we missplayed and then use the information to adjust our world model of the marimba. But maybe this is not needed and a simple score suffices for us

@ateRstones The mismatched note and final score(recall) is available now from this issue #201, just haven't been merged into the main branch. I will create a PR when I am done with its visualization.

Juphex, 2023-09-29:
can we close this @Flova ?

Juphex, 2023-09-29:
I close this. Due to no interactions and probably out of our scope.

closed 2023-09-29 by Juphex