UHHRobotics22-23/MarimbaBot #182: additional commands

scaredycode, 2023-07-06:
- play in specific tempo -> "Marimbabot play in 120 bpm"
- play in specific loudness -> "Marimbabot play in forte"
- increase/decrease Tempo by specific value -> "Marimbabot play faster by 60 bpm"
- increase/decrease loudness by specific value -> "Marimbabot play louder by 0.2"
- all "play" commands also as "preview" -> "Marimbabot preview", "Marimbabot preview slower by 30 bpm", "Marimbabot preview in 80 bpm", ...

Juphex, 2023-07-09:
All commands are here https://github.com/UHHRobotics22-23/marimbabot/blob/main/marimbabot_behavior/src/marimbabot_behavior/behavior_node.py, right?

Maybe something like play everything a half-note / note higher? :)

scaredycode, 2023-07-13:
> All commands are here https://github.com/UHHRobotics22-23/marimbabot/blob/main/marimbabot_behavior/src/marimbabot_behavior/behavior_node.py, right?

What do you mean by "are here" exactly? Currently the behavior can work with the commands that are marked as "already supported in behavior". However, these commands are not supported by the speech module yet, meaning while we theoretically can work with the instruction, we cant recognize it in the speech detection layer.
The issue we face right now is the recognition of not only a given text but a pattern instead ("play in <x> bpm"

yunlong-wang-cn, 2023-09-14:
new command: **`play/preview in a loop`**

old command:
play in **specific tempo** -> "Marimbabot play in 120 bpm" - already supported in behavior
play in **specific loudness** -> "Marimbabot play in forte"
**increase/decrease Tempo** by specific value -> "Marimbabot play faster by 60 bpm" - already supported in behavior
**increase/decrease loudness** by specific value -> "Marimbabot play louder by 20 percent"
all "play" commands also as **"preview"** -> "Marimbabot preview", "Marimbabot preview slower by 30 bpm", "Marimbabot preview in 80 bpm", ... - already supported in behavior

Juphex, 2023-10-10:
Can we close this @scaredycode  @yunlong-wang-cn  ?

yunlong-wang-cn, 2023-10-10:
yes, all command stuff settle down.

closed 2023-10-14 by Flova