UHHRobotics22-23/MarimbaBot #184: Modify the read lilypond sequence

scaredycode, 2023-07-06:
- [ ] increase/decrease tempo by fixed value
- [ ] increase/decrease tempo by specified value
- [ ] set tempo to specified value
- [ ] increase/decrease volume by fixed value
- [ ] increase/decrease volume by specified value
- [ ] set volume to specified value
- [ ] add swing
- [ ] transpose
- [ ] ...

Juphex, 2023-09-29:
Are the other ones in progress @scaredycode ?

ateRstones, 2023-10-02:
#212 seems to be included in this ticket? I'm wondering @scaredycode

scaredycode, 2023-10-03:
> Are the other ones in progress @scaredycode ?

no, @yunlong-wang-cn and I decided against including anymore lilypond modification functionality.
I tested adding swing and transpositions some weeks ago, but this of course caused the parser to fail :)

Regarding setting volume to a specified value, I suggested adding commands such as "Play in fortissimo" or similar, but it is currently not implemented neither in the speech node nor in the behavior. We also thought that this might not be a use case a "usual" user would go for (more "Play x steps louder" might be much more intuitive).

Juphex, 2023-10-10:
@scaredycode 
I close this because you moved this into "Done".