UHHRobotics22-23/MarimbaBot #196: Refactor marimbabot_bringup launch files

Juphex, 2023-07-13:
E.g. combine all launch files

marimbabot.launch(File) | add behavior to bringup launch file |  
-- | -- | --
 

[marimbabot.launch](https://github.com/UHHRobotics22-23/marimbabot/blob/main/marimbabot_bringup/launch/marimbabot.launch)
(File)
	
[add behavior to bringup launch file](https://github.com/UHHRobotics22-23/marimbabot/commit/1b297b0d432a0468995001f3eff6b1cf1bd3a5ac)
	
[marimbabot_ur5_bringup.launch](https://github.com/UHHRobotics22-23/marimbabot/blob/main/marimbabot_bringup/launch/marimbabot_ur5_bringup.launch)
(File)
	
[two mallet configuraitons for motion (](https://github.com/UHHRobotics22-23/marimbabot/commit/42109351b0ebbef6e1fe07df9c86935c50dc58cf)https://github.com/UHHRobotics22-23/marimbabot/pull/176[)](https://github.com/UHHRobotics22-23/marimbabot/commit/42109351b0ebbef6e1fe07df9c86935c50dc58cf)
	
[marimbabot_ur5_bringup_with_servo.launch](https://github.com/UHHRobotics22-23/marimbabot/blob/main/marimbabot_bringup/launch/marimbabot_ur5_bringup_with_servo.launch)
(File)
	
[two mallet configuraitons for motion (](https://github.com/UHHRobotics22-23/marimbabot/commit/42109351b0ebbef6e1fe07df9c86935c50dc58cf)https://github.com/UHHRobotics22-23/marimbabot/pull/176[)](https://github.com/UHHRobotics22-23/marimbabot/commit/42109351b0ebbef6e1fe07df9c86935c50dc58cf)

Flova, 2023-07-14:
What do you mean by combine all launch files?

Juphex, 2023-09-06:
@Flova 
I am not quite sure what the state of the launch files were.
But they seem to be fine, except maybe we add some documentation

closed 2023-09-06 by Juphex

Flova, 2023-09-06:
Does the main lauch file launch everything?

Flova, 2023-10-03:
It seems so in yesterdays tests