UHHRobotics22-23/MarimbaBot #202: sound_play_node is died

yunlong-wang-cn, 2023-07-20:
[soundplay_node-5] process has died [pid 125213, exit code 1, cmd /opt/ros/noetic/lib/sound_play/soundplay_node.py 
    __name:=soundplay_node __log:=/home/wang/.ros/log/524c6e1e-2718-11ee-8ff8-b51692687720/soundplay_node-5.log].
    log file: /home/wang/.ros/log/524c6e1e-2718-11ee-8ff8-b51692687720/soundplay_node-5*.log

When I start the marimbabot_audio, it said the sound_play_node is dead.

Juphex, 2023-07-28:
> When I start the marimbabot_audio, it said the sound_play_node is dead.

@yunlong-wang-cn  What is the branch? Is starting the marimbabot.launch of marimbabot_audio enough to reproduce the issue?

yunlong-wang-cn, 2023-07-28:
> Is there any previous output?

@Flova  It said some wrong with gstreamer.

yunlong-wang-cn, 2023-07-28:
> > When I start the marimbabot_audio, it said the sound_play_node is dead.
> 
> @yunlong-wang-cn Which branch? Is starting the marimbabot.launch of marimbabot_audio enough to reproduce the issue?

main branch. Actually, I thinks just launch the marimbabot_audio/marimbabot.launch is enough to reproduce the problem.

yunlong-wang-cn, 2023-08-03:
I think this is the problem with the sound_play node, this is the full message:

started roslaunch server http://wang-Y7000P:34601/

SUMMARY
========

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.16.0

NODES
  /
    soundplay_node (sound_play/soundplay_node.py)

auto-starting new master
process[master]: started with pid [37764]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to a20c060c-31ff-11ee-9c0a-dfdc9ded8337
process[rosout-1]: started with pid [37785]
started core service [/rosout]
process[soundplay_node-2]: started with pid [37788]

**************************************************************
Error opening pygst. Is gstreamer installed?
**************************************************************

[soundplay_node-2] process has died [pid 37788, exit code 1, cmd /opt/ros/noetic/lib/sound_play/soundplay_node.py __name:=soundplay_node __log:=/home/wang/.ros/log/a20c060c-31ff-11ee-9c0a-dfdc9ded8337/soundplay_node-2.log].
log file: /home/wang/.ros/log/a20c060c-31ff-11ee-9c0a-dfdc9ded8337/soundplay_node-2*.log

yunlong-wang-cn, 2023-08-03:
@scaredycode  I think you working on the sound_play node, right? please take a look.

scaredycode, 2023-08-03:
> @scaredycode  I think you working on the sound_play node, right? please take a look.

I have worked with it successfully some weeks prior but not recently. I will look into it and see if I can reproduce the error 👍

yunlong-wang-cn, 2023-08-16:
when I launch: `roslaunch sound_play soundplay_node.py`, the same error occurs. Perhaps I have something to install or some configuration of the environment.

Juphex, 2023-08-17:
> `roslaunch sound_play soundplay_node.py`

That is the wrong command.
The right (and working ) one would be `roslaunch sound_play soundplay_node.launch`

> > > When I start the marimbabot_audio, it said the sound_play_node is dead.
> > 
> > 
> > @yunlong-wang-cn Which branch? Is starting the marimbabot.launch of marimbabot_audio enough to reproduce the issue?
> 
> @Juphex main branch. Actually, I thinks just launch the marimbabot_audio/marimbabot.launch is enough to reproduce the problem.

I cannot reproduce this. When I run `roslaunch marimbabot_audio marimbabot.launch` I get :
`[ERROR] [1692261519.384082798]: gstreamer: Could not open audio device for recording.
`
but this is probably because I have not recording device (set up).
`rosnode list` also shows that the sound_play node runs.

Furthermore when I run `rostopic pub /robotsound sound_play/SoundRequest "{sound: -3, command: 1, volume: 1.0, arg: 'Hello its me', arg2: ''}"
` everything is fine. Sound is output as expected.

Maybe you check the rosdeps. @yunlong-wang-cn

closed 2023-08-17 by Juphex