UHHRobotics22-23/MarimbaBot #206: Final presentation at the latest in the first weeks of the semester

ateRstones, 2023-08-07:


ateRstones, 2023-08-07:
I was thinking about using google presentations for this, as it allows for easy collaborative work. Other thoughts about this?

Flova, 2023-08-29:
What are the requirements for the final presentation? Are sildes required? Is it a public demo?

ateRstones, 2023-09-05:
I think the last years were a public demo. Please ask Michael or Norman

ateRstones, 2023-10-02:
Correction, we don't necessarily need a presentation but have to only to a demo with potentially slides as support

cvhex, 2023-10-05:
https://docs.google.com/presentation/d/1Oi_m2aeNIBA3nVfwhzWw8N8VOkm-Lhm6JC1UM5gr76Yg/edit?usp=sharing
The Link for the presentation

Flova, 2023-10-05:
The link seems to be broken

cvhex, 2023-10-12:
https://docs.google.com/presentation/d/1Oi_m2aNIBA3nVfwhzWw8N8VOkm-Lhm6JC1UM5gr76Yg/edit?usp=sharing

Flova, 2023-10-16:
We should probably delete this public link with editing permissions.

berkgungor, 2023-10-17:
DOne