UHHRobotics22-23/MarimbaBot #209: pattern  of command recognition

yunlong-wang-cn, 2023-08-10:
@scaredycode 
Consider adding the pattern for the command recognition

yunlong-wang-cn, 2023-08-10:
@scaredycode Sorry, I can not find which pattern we need to recognize for command recognition, I think you mentioned it somewhere, could you show me again?

scaredycode, 2023-08-10:
> @scaredycode Sorry, I can not find which pattern we need to recognize for command recognition, I think you mentioned it somewhere, could you show me again?

 I talked about such pattern commands in this issue here:
https://github.com/UHHRobotics22-23/marimbabot/issues/182

yunlong-wang-cn, 2023-08-22:
@scaredycode  I haven't used the LLM to build the command extraction, it is too huge,  can not be run in the PC near UR5, so I write a much single logic with regular expression, and it work pretty goods with the pattern above.

Flova, 2023-08-29:
Is this issue also covering robustness against things like `marimbabot plays`, `marimbabot, plays`, ...?

yunlong-wang-cn, 2023-09-14:
@scaredycode For your PR#242, give you info from this PR#218. 
Here give you the list of all possible commands in the format of `{"action": "increase speed", "parameter": "20bpm"}`.

The following "parameter" is always changeable and could be an empty string to denote user hasn't given any parameters of this action.

1. For speed: 
 `{"behavior": "play", "action": "increase speed", "parameter": "20bpm"}`.
 `{"behavior": "play", "action": "decrease speed", "parameter": "40bpm"}`
 `{"behavior": "play", "action": "setup speed", "parameter": "20bpm"}`
 2. For volume:
  `{"behavior": "play", "action": "increase volume", "parameter": "20bpm"}`.
 `{"behavior": "play", "action": "decrease speed", "parameter": "40bpm"}`
 `{"behavior": "play", "action": "setup speed", "parameter": "20bpm"}`
3. other:
"play" of behavior can be replace by `preview` or `loop`

yunlong-wang-cn, 2023-09-28:
@scaredycode FYI