UHHRobotics22-23/MarimbaBot #234: inconsistency in vision node

berkgungor, 2023-08-17:
- Hand written tempo and loudness values (tempo=120 , fff, pp) have considerable consistency.
- Recognition of magnetic notes is good but has a little inconsistency.

berkgungor, 2023-08-17:
[image]

Juphex, 2023-08-17:
related to 
https://github.com/UHHRobotics22-23/marimbabot/issues/43
https://github.com/UHHRobotics22-23/marimbabot/issues/66

Flova, 2023-08-29:
As they are in the beginning of the sequence we can also collect data only featuring tempo information, but no notes. This should be way faster to collect.

Flova, 2023-09-21:
The same procedure is possible for keys etc

Juphex, 2023-09-27:
What is the status of major/minor keys + bpm collection from last week @cvhex ?

Juphex, 2023-10-10:
Can we close this @Flova ?

closed 2023-10-10 by Flova