UHHRobotics22-23/MarimbaBot #235: [Command Recognition] Need to collect more data for keyword spotting

yunlong-wang-cn, 2023-08-17:
Since currently the model is built totally by my voice, it has a strong bias.  So we need to collect more dataset:
1. More female voice.
2. More other people's voices, perhaps 20 `.wav` file for the keyword `Hi, marimbabot` is enough.

How to do it:
1. Refer to this installation [instruction](https://github.com/MycroftAI/mycroft-precise#installation) to install Mycroft.
2. Refer to this [instruction](https://github.com/MycroftAI/mycroft-precise/wiki/Training-your-own-wake-word#recording-samples) for data collection.(Don't forget to activate the venv.)
3. By recording the keyword for waking, each `.wav` file should contain once keyword.
4. And also you can record some no-wake-keyword, which should be some keywords similar to the wake_keyword, such as "hi", and "robot", "marimba". All of those words word shouldn't independently wake the robot.

yunlong-wang-cn, 2023-08-22:
@scaredycode  if you have time, I need you help to test if this part works well for your voice

Flova, 2023-08-29:
I will do this in person next time in the lab :)

yunlong-wang-cn, 2023-09-05:
currently, the following voices are collected, waiting for further testing, and decide if more collection.

[image]

Juphex, 2023-10-04:
We definitely need to collect my voice as it was not recognized 😅
@yunlong-wang-cn

yunlong-wang-cn, 2023-10-04:
yes

yunlong-wang-cn, 2023-10-12:
close as cpmplete