UHHRobotics22-23/MarimbaBot #237: [Behavior Node] Dummy client for send plan to motion planning

yunlong-wang-cn, 2023-08-17:
[image]
For using the rosbag to collect some dataset to calibrate the audio processing,  we need a dummy client for how to let robot play a custom music sequence and feddback the music plan, and then can forward it to audio node with absolute plan.

Flova, 2023-08-29:
Why do we need a dummy client and not a cli publisher (`rostopic pub`)

Juphex, 2023-08-31:
I think he needed it for testing.

yunlong-wang-cn, 2023-08-31:
yes, just for testing

Juphex, 2023-09-06:
So, should we forfeit this topic/issue?
@yunlong-wang-cn 
@Flova

Flova, 2023-09-11:
I don't really see a benefit over the standard Ros cli tbh.

Juphex, 2023-09-14:
I close this then @yunlong-wang-cn