UHHRobotics22-23/MarimbaBot #240: problem on faster / slower options for tempo

berkgungor, 2023-08-17:
Observed that after " marimbabot play faster" and "marimbabot play slower" commands, tempo was not updated properly. Robot speed has not changed and same tempo value is printed from behavior node instead of updated value.

yunlong-wang-cn, 2023-08-17:
One guess is that the command is not detected, in the current version, the command is just a string match, if the template is "play faster", then if you say "play**s** faster", it will not be detected.

scaredycode, 2023-08-17:
The issue was that the vision model detects tempo as "\tempo 4=120" but the behavior expects "\tempo 4 = 120". The parser works for both versions, but we should of course go with the syntax of our model.
I fixed the issue in the behavior in my most recent PR (241) which also adds volume adjustments to the behavior 👍

Juphex, 2023-09-29:
Is this related to https://github.com/UHHRobotics22-23/marimbabot/issues/258 ?

Flova, 2023-09-29:
It worked the last time I tested it.