UHHRobotics22-23/MarimbaBot #245: [CI] how to add apt install in the CI?

yunlong-wang-cn, 2023-08-22:
Since some software is necessary for `precise-runner`,  therefore, before `pip install precise-runner`, we need to make sure the following software exist:

    sudo apt install python3-pip \
    libopenblas-dev \
    python3-scipy \
    cython \
    libhdf5-dev \
    python3-h5py \
    portaudio19-dev

My problem is how to put this apt installation into CI procedure, otherwise the CI can not be successed.

ateRstones, 2023-08-28:
I think you should be able to modify the install depencies step in https://github.com/UHHRobotics22-23/marimbabot/blob/main/.github/workflows/main.yml

Flova, 2023-08-29:
All of them are available via rosdep, so you can add a dependency to your `package.xml` and need no other changes. Looking at the package index at https://github.com/ros/rosdistro/blob/master/rosdep/base.yaml we see the following:

libopenblas-dev: Available as [libopenblas-dev](https://github.com/ros/rosdistro/blob/a691abf5ef31dfa719ba6b7f08bd6eed858581b7/rosdep/base.yaml#L4376C1-L4376C16)
cython: Avaible as [cython](https://github.com/ros/rosdistro/blob/a691abf5ef31dfa719ba6b7f08bd6eed858581b7/rosdep/python.yaml#L162)
libhdf5-dev: Available as [hdf5](https://github.com/ros/rosdistro/blob/a691abf5ef31dfa719ba6b7f08bd6eed858581b7/rosdep/base.yaml#L1870)
python3-h5py: Available as [python3-h5py](https://github.com/ros/rosdistro/blob/a691abf5ef31dfa719ba6b7f08bd6eed858581b7/rosdep/python.yaml#L6286)
portaudio19-dev: Available as [portaudio19-dev](https://github.com/ros/rosdistro/blob/a691abf5ef31dfa719ba6b7f08bd6eed858581b7/rosdep/base.yaml#L7006)

yunlong-wang-cn, 2023-09-04:
@Flova  What is the different between the `base.yaml` and `python.yaml` in rosdistro?

Juphex, 2023-09-04:
Hey @yunlong-wang-cn .
You can take a look at the logs https://github.com/UHHRobotics22-23/marimbabot/actions/runs/6075807070/job/16482709856?pr=230#step:3:847 . They show, that `  $ ( source /opt/ros/noetic/setup.bash && rosdep install -q --from-paths /root/target_ws/src --ignore-src -y | grep -E '(executing command)|(Setting up)' ; )`  is executed before python requirements are installed.

Juphex, 2023-09-04:
Hey @yunlong-wang-cn,
I could reproduce your error. 
I just tried in another branch - as soon as I have fixed it (hopefully) I create a PR that bases on your branch so you can merge it into that.

Juphex, 2023-09-04:
Here it is https://github.com/UHHRobotics22-23/marimbabot/pull/253

Flova, 2023-09-05:
We could just integrate the python stuff in the main CI workflow this should resolve the need for an extra rodep install.

Flova, 2023-09-05:
Nvm I saw your PR. Your solution is faster but a bit wastefully because of the two rosdep installations.

yunlong-wang-cn, 2023-09-05:
currently, the CI works well.  And I prefer the second one, which could save some time for us.

closed 2023-09-06 by Juphex