UHHRobotics22-23/MarimbaBot #250: Adjust Timing information that is returned by planning

Juphex, 2023-08-31:
Current:
It returns a list of https://github.com/UHHRobotics22-23/marimbabot/blob/main/marimbabot_msgs/msg/HitSequenceElement.msg .

It makes more sense to send the calculated time when the first note of a sequence is hit and then send it to the behavior which sends it to audio.
Why?: The audio has the ground  truth note sequence and wants to compare with it -> Whether the played notes are accurate or not. It does not make sense to send the timings for each note because we should orient on the note sequence that *should* be played and not the one what was *actually played* because it might be wrong/incorrect.

@Flova 
@yunlong-wang-cn 

Therefore, the audio should calculate the timings for each note based on the time when the first note was hit and add offsets in order to calculate the following note times.

yunlong-wang-cn, 2023-09-06:
@Juphex 
I have some questions:
1. What do you mean by "the note sequence that should be played ", Does that mean a plan? if so,  who should generate this plan? And what is the content this plan has?
2. For "the audio should calculate the timings for each note based on the time when the first note was hit and add offsets in order to calculate the following note times.",  have you ever thought about what could happen if the first hit is missed? So for this purpose, I think it is better to give an absolut timing.

Juphex, 2023-09-06:
You're absolutely right, the "note sequence that should be played" refers to the expected or ground truth sequence of notes, and the planned sequence is how we intend to play them. Ideally, both of these sequences should align, and that's the goal of the audio component, to compare them and provide feedback, correct?
Otherwise we would compare sort of with ourselves instead of a correct interpretation of the note sequence.

So, it makes sense to provide the audio with only the timestamp of the first note. The rest of the timestamps can be calculated based on the Lilypond sequence (i.e., the expectation).

Even if the first hit doesn't go as planned (but we have the planned time), we still have the information about when the other notes should be played. Theoretically, we can compare for the remaining notes. Let's revisit this tomorrow for further discussion, shall we? @Flova @yunlong-wang-cn

yunlong-wang-cn, 2023-09-06:
@Juphex ok, see you tomorrow.

yunlong-wang-cn, 2023-09-07:
@Juphex 
So I think you want to let the behavior node generate the "plan" aka. timestamp. 
But I have a concern, since we use many constraints in motion planning, if we let the behavior node generate the timestamp, there will be more constraints for motion planning, and it could fail a lot, which means it is really hard to find a feasible trajectory plan. So for this concern, we need to ask @Flova.

I know the current timestamp is a little bit tricky, but we to make sure our crews can finish all the stuff before DDL.

yunlong-wang-cn, 2023-09-07:
During the meeting with @Juphex , we achieved the following consensus:

The behavior node will send the "hit plan" (aka. expectation, as mentioned above) to the audio node when the robot finished it's last motion. The time should still be the absolute time.

For the audio node, the music note recognition is always on, when it receives the "hit plan" from the behavior node, it will retrieve the music note from the records according to the time windows of that "hit plan".

Please check if my understanding is right. @Juphex

Juphex, 2023-09-07:
Yes,
it it just about how we generate the HitSequences the audio is comparing with

closed 2023-10-02 by Flova