UHHRobotics22-23/MarimbaBot #263: add documentation on planning node

Juphex, 2023-09-27:
Please, someone document how to use scripts and test/run the planning node.
Preferably in the README of the package. This also helps others how to use this package.
This may include changing the "main" README.
These steps are needed:

roslaunch marimbabot_ur5_flex_double_moveit_config demo.launch 
rosrun marimbabot_planning planning 

and running any of the scripts, e.g.:
rosrun marimbabot_planning sound_pad.py

Flova, 2023-09-28:
Please use the launch file and not `rosrun` if one is present as it might perform additional things like parameter loading

berkgungor, 2023-10-05:
how we are gonna make specifically py files run without using `rosrun` ?

Flova, 2023-10-05:
What do you mean? `rosrun` is a valid option no launch file exists.

berkgungor, 2023-10-06:
what did you mean while saying "Please use the launch file and not rosrun" ?, bcs i added rosrun commands for .py files into the planning documentation.

Flova, 2023-10-07:
This text of this issue states:

> roslaunch marimbabot_ur5_flex_double_moveit_config demo.launch
rosrun marimbabot_planning planning

Which should be

>roslaunch marimbabot_ur5_flex_double_moveit_config demo.launch
roslaunch marimbabot_planning marimbabot.launch

Flova, 2023-10-17:
Done

closed 2023-10-17 by Flova