UHHRobotics22-23/MarimbaBot #274: Specify volume

scaredycode, 2023-10-03:
(might be a bit late, but it has been addressed last week so I will open the issue anyways)

Currently we can  control the volume (only) with the louder/softer commands.
We can specify the tempo by giving the command "Play in x bpm".
Idea: Do a similar thing for the volume: "Play in fortissimo".

We support 8 different volume "levels" in our lilypond sequences that could be directly addressed in the voice command:

pronunciation | music notation | meaning
-- | -- | --
fortississimo | fff | very very loud
fortissimo | ff | very loud
forte | f | loud
mezzo-forte | mf | moderately loud
mezzo-piano | mp | moderately quiet
piano | p | quiet
pianissimo | pp | very quiet
pianississimo | ppp | very very quiet

Flova, 2023-10-03:
While testing the louder/softer commands yesterday I noticed them having no effect. Maybe I did something wrong, but adding fff to the vision data and telling the robot to play louder or softer didn't change the commanded notes. Also using fortississimo as a default might be reasonable, as the robot is not playing that loud and playing with a medium loudness by default might confuse people as they think it is the maximum if they dont care or explicitly specify a loudness.

berkgungor, 2023-10-03:
when I tested the volume last week, we observed clear changes by ear as well as note.velocity value in interpreter.py. 
The reason that robot does not play with maximum loudness level (127) when the command was fff (maximum loudness), is because @scaredycode also added "mercado" case as highest volume. In short, we can play with ~95/127 loudness for fff but robot only play with 127/127 when the command is "mercato". 
On the other hand, specifying the volume by commands like above looks much better option to observe the changes in volume level.

scaredycode, 2023-10-05:
> While testing the louder/softer commands yesterday I noticed them having no effect. Maybe I did something wrong, but adding fff to the vision data and telling the robot to play louder or softer [...]

I identified the problem. The default volume assignment was implemented but never used. I fixed it here https://github.com/UHHRobotics22-23/marimbabot/pull/280

Juphex, 2023-10-10:
I assume this can be closed.
@scaredycode @Flova @yunlong-wang-cn @berkgungor

closed 2023-10-10 by Juphex

yunlong-wang-cn, 2023-10-10:
sure